Duet3D/DuetWebControl #1: Languages.xml problem

gonAlonso, 2015-05-29:
Hi Chrishamm,
I would like to translate you UI to different languages, but I've found it impossible. The point is, I modified your original languages.xml to fit Spanish location, but when I update it to the Duet, it doesn't show any languaje. Is it possible to add another languaje to this file?
Besides, it the UI takes XML files as system files, it's uploaded to /sys instead to /www
Thank you for this fantastic UI!

chrishamm, 2015-05-29:
Hi gonAlonso,

It's possible that the language file isn't properly uploaded to the /www directory, in that case I'll have a look at the code again. Meanwhile you could verify this by connecting to your Duet via FTP: If language.xml is present in /sys, you won't see all the languages from the translation file. Then you must move it to /www, reload the web interface once and you should be able to change the language on the settings page as you'd expect.

When your translation file is complete, I'd appreciate if you shared it - I could add it to the next version of my web interface if that's okay for you.

gonAlonso, 2015-05-29:
That's precisely what I did. I removed the SD card from the Duet and checked it on my computer. So I moved it to the right place and reloaded the UI after the card was in the Duet again. No success. The UI shows _English_ only on the drop-down list on the _Settings_ page
What I did was to copy the whole _&lt;de name='Deusch'&gt;...&lt;/de&gt;_ , I modified it to be _&lt;es name='Espa&amp;nacute;ol'&gt;..&lt;/es&gt;_ and edited all items.
How can I send the file to you?

gonAlonso, 2015-05-29:
Hei, I've found the problem!
It was a typo, my fault.
Now, I've realized some texts need to be polished to better suit its intended purpose. Let me know how to share it with you, I'll send it as soon as I feel it's done
Thank you

closed 2015-06-10 by gonAlonso