Duet3D/DuetWebControl #7: Webcam Integration

nokianhunter, 2016-01-17:
Hi there,

I have a problem with the Webcam URL I enter in the DuetWebControl settings.
It needs to point to an image file. So I tried "http://192.168.0.35:8080/photo.jpg", while running the IP Webcam App on Android. (Checked with browser and it works there)
But the Print Status Page always shows "image not available".
Any hint is appreciated.

Regards
Nokian

chrishamm, 2016-01-17:
I believe there is a typo in viewmodel.js. Could you open that file, go to line 344 and replace

if (newURL.indexOf("?") != -1) {

with

if (newURL.indexOf("?") == -1) {

and check if that fixes your problem?

nokianhunter, 2016-01-17:
Well, that did the trick.
Thanks for your fast response!

closed 2016-01-17 by nokianhunter