Duet3D/DuetWebControl #9: no picture in Webcam option

criticallimit, 2016-01-26:
Like the other item with the rtsc stream,
my stream isn´t shown as well.
This is my url to the cam:

http://192.168.2.27:88/cgi-bin/CGIStream.cgi?cmd=GetMJStream&usr=Webcontrol&pwd=+++++

criticallimit, 2016-01-26:
If I put the stream direct to the img src it is working. But than the window is not hideable in the webinterface...

<div id="panel_webcam" class="panel panel-default hidden">
									<div class="panel-heading">
										<a id="a_webcam" href="#">Webcam Surveillance</a>
										<span id="span_webcam" class="glyphicon glyphicon-menu-down"></span>
									</div>
									<div class="panel-body">
										[image]
								
									</div>
								</div>

chrishamm, 2016-02-07:
I guess it is the same problem as this one: https://github.com/chrishamm/DuetWebControl/issues/7 Can you check that?

I will fix the described problem in my next DWC version.

criticallimit, 2016-02-22:
didn´t changed the problem...

criticallimit, 2016-02-22:
update:

working now. Forgot to clear the browsers cache.

chrishamm, 2016-03-02:
This has been fixed in v1.09

closed 2016-03-02 by chrishamm