Duet3D/DuetWebControl #12: Help system for DWC

PRouzeau, 2016-06-09:
Hello Christian, I left you a message a week ago in the RepRap forum private box about an help system for DWC that I am developing, but this message is still unread after one week, could you please have a look and tell me how to proceed?

Regards, Pierre

chrishamm, 2016-12-28:
Hi Pierre, have you had a chance to test your help system on a Duet WiFi yet? AFAIR it's working on the wired Duet, but I haven't tested this on a Duet WiFi yet. Also if I merge in your changes, I'd like to be able to create two packages in the future: A core version that ships DWC as-is and another one with your Aquilegia help system.

Hence I've been planning to modifying the build script to apply a patch to the core files and to package everything as separate bundles.

PRouzeau, 2016-12-28:
Hello Christian,

I was sent a white Duet WiFi board for tests a few month ago and tested 
it ok. The version yet on GitHub is working on Duet WiFi and there are a 
few comments in the release notes. See also in the help itself for Duet 
WiFi build with Windows 10. I got troubles in minifying so I modified 
the build procedure accordingly.

However, there are a few points:

- I have not worked on it recently on it and I may not be able to do a 
lot of work for a few month to come, due to other priorities

- The Github version works, but it is not the last version

- I implemented and tested a direct load from the interface both for the 
program and the help files (not on line) (by an interface button)

- In the non published version, I set the whole help load local optional 
(say, if not loaded, it looks on the web)

- Maintaining two different help systems is not quite sustainable on the 
long term and I no longer update the help files, even if I I have done a 
conversion routine from Wikimedia markup to Aquilegia markup to help me.

- I think schematics are key in an help documentation for such product 
and I put a lot of hours in developing schematics, but they were not 
reused, so why continue ?

- I had to abandon Windows 10 anniversary (too many bugs) and I no 
longer have a bash shell available (I didn't rebuild Linux setup) , so I 
cannot yet rebuild for the Duet WiFi.

Regards, Pierre

mydevpeeps, 2020-08-25:
This is really old, and yet still open. Is this something we could pick back up again?