Duet3D/DuetWebControl #15: Finding supported webcams

BrianGilbert, 2016-08-10:
Hi Chris

Are there any specifics we need to know when trying to find a suitable webcam?

Cheers
Brian

chrishamm, 2016-08-11:
Hi Brian,

At the moment only images are supported, so you need to make sure your webcam can deliver screenshots rather than a video live stream. If you find a camera that supports HTML5 video streaming, I could add support for it too, but I don't want to use Flash or other plugins to provide this functionality.

Cheers,
Christian

OricTosh, 2016-08-11:
For your information 
My printer set-up uses OctoPrint upon a Pi which runs a MPEG streamer from the attached Pi Camera.
I have entered the following into the Webcam Intergration box 
http://raspberry's IP/webcam/?action=stream which provides a good view within Web Control Panel

closed 2016-12-28 by chrishamm