Duet3D/DuetWebControl #17: System Editor not working with Safari

tomasf, 2016-08-28:
Saving edited files using the System Editor doesn't work with Safari. Here's a relevant StackOverflow thread: http://stackoverflow.com/questions/33821631/alternative-for-file-constructor-for-safari

lignumaqua, 2016-10-31:
The latest version of Safari (10.0.1) implements the file() constructor which solves the problem for Safari. Unfortunately IE and Edge have the same issue and have no support for the file() constructor.

The following code _may_ work in IE and Edge - I haven’t tested it in this case, but it's worked for me elsewhere. I don’t have the environment to build this app so am unable to test it fully.:

Replace line 22 in upload.js

`var file = new File([content], filename, { type: "application/octet-stream" });`

With

```
var file = new Blob([content], { type: "application/octet-stream" });
file.lastModifiedDate = new Date();
file.name = filename;
```

Ideally this would only be done if the browser is detected as IE or Edge, or if new File() isn’t valid.

chrishamm, 2016-11-03:
Thanks for the work-around, I've just tested it and changed the upload code to use Blob instead of File. These changes will be available once DWC 1.13 has been released.

closed 2016-11-03 by chrishamm

lignumaqua, 2016-11-05:
Chris - you left out the file.name line in 1.13, that's breaking the config.g reboot test

`file.name = filename;`