Duet3D/DuetWebControl #25: Models sliced with simplify 3d show incorrect details

schneik80, 2017-01-26:
every model sliced in simplify3d shows incorrect details in web view.
all models show 225 which is my configured max Z
all models show layer x of 1126

[image]

schneik80, 2017-01-26:
gcode attached
[marvin.gcode.zip](https://github.com/chrishamm/DuetWebControl/files/731326/marvin.gcode.zip)

chrishamm, 2017-01-26:
Both your start and end G-code sections contain absolute Z movements and the firmware reads these Z coordinates, that's why the web interface displays wrong values.

As a work-around you can put your start G-code into a macro file and call it via M98. If you insist on moving Z to the same position when the print is complete, do the same with your end G-code or append "; End" to the last G1 Z code. Alternatively you can replace the absolute Z movement with a relative movement by encapsulating the last G1 code with G90/G91 like:

G91 ; relative movement
G1 Z5 ; lift Z by 5mm
G90 ; absolute movement

closed 2017-03-03 by chrishamm