Duet3D/DuetWebControl #28: Flatten the UI

iDeveloUK, 2017-03-06:
Would you like me to go through and remove all the shadowing and make the UI flat?

chrishamm, 2017-03-06:
If you want to make the UI flat, just remove the CSS link which loads bootstrap-theme.css. I for one like DWC better with the shadows and gradients, so I haven't put much thought in this topic yet.

If you like I could make this option variable in DWC 1.16.

BrianGilbert, 2017-03-06:
I'd like to see a screenshot of the difference, but if it's simple.. +1 on the option

iDeveloUK, 2017-03-07:
This is with the gradients and border radius removed.
[image]

iDeveloUK, 2017-03-07:
And this: 

[image]

iDeveloUK, 2017-03-07:
Or something to take part of that boot-strappy feel away
[image]

chrishamm, 2017-04-19:
OK, I'll change that behavior in my next version. If you want me to add a new theme (via a new CSS file), send it over and I'll add it.

closed 2017-04-20 by chrishamm