Duet3D/DuetWebControl #30: Build Instructions

iDeveloUK, 2017-03-07:
So the build script reads the following:

<link href="css/animate.css" rel="stylesheet">
<link href="css/bootstrap.min.css" rel="stylesheet">
<link href="css/bootstrap-theme.min.css" rel="stylesheet">
<link href="css/bootstrap-slider.min.css" rel="stylesheet">
<link href="css/defaults.css" rel="stylesheet">

It then puts all the css into one file and minifys it. Why does reprap.htm not contain the non min version of the stylesheets? I would have presumed you would edit bootstrap-theme.css, bootstrap.css etc and then let the script do it's work?

What is your editing process? are you using the supplied build script?

chrishamm, 2017-03-07:
Well, the build script minifies each CSS file and then compresses the result via gzip because it allows us to save space which speeds up the page loading. I found it quite useful being able to look into the source files when I make changes, though.

Besides I don't want to edit supplied 3rd-party files because that lets me easily update those components if I need to. And yes, of course I use the build script to compile new versions.

iDeveloUK, 2017-03-07:
The initial code was stripped out in my first comment!

The stylesheets it reads are these:

`              <link href="css/animate.css" rel="stylesheet">

		<link href="css/bootstrap.min.css" rel="stylesheet">

		<link href="css/bootstrap-theme.min.css" rel="stylesheet">

		<link href="css/bootstrap-slider.min.css" rel="stylesheet">

		<link href="css/defaults.css" rel="stylesheet">`

And that's specifying the minified version already. When you edit the css don't you edit the un-minified version because currently it's packing the minified code which makes it hard to edit if needed?

`- Minifying core/css/animate.css...
- Minifying core/css/bootstrap.min.css...
- Minifying core/css/bootstrap-theme.min.css...
- Minifying core/css/bootstrap-slider.min.css...
- Minifying core/css/defaults.css...`

In theory you don't need the minified version of the css in the css folder and only one copy of each css file will exist.

I understand you don't need to edit the main bootstrap css but the theme css could be edited if needed. OR is default.css used as the edit's file?

chrishamm, 2017-03-07:
Ahh yes, now I see what you mean. Good point, you're right - in fact it no longer makes sense to include these minified files in the HTML file. I haven't removed them yet because the first DWC versions were not available for the Duet WiFi, hence the build script wasn't needed and I only had to compress the whole folder. I'll change that when I get the chance.

chrishamm, 2017-03-07:
PS: defaults.css is only used for my own CSS customisations

chrishamm, 2017-04-19:
Redundant files will be removed when 1.15b is released.

closed 2017-04-19 by chrishamm