Duet3D/DuetWebControl #32: Download all config files with one click?

yonkiman, 2017-03-14:
Great software!  I thought I'd be missing Octoprint but I'm quite happy with DuetWebControl so far.

One suggestion though.  I'm trying to backup all my .g files, and the process takes a lot of clicks...  Maybe there's an easier way that I haven't discovered, but here's how I'm doing it now:

- Click on Settings
- Click on System Editor
- Click download for the file I want to download
- Right click "Save As"
- Change "Save as type" to "All Files" (prevents .txt appending to filename)
- Replace "rr_download.txt" with correct filename (like "homex.g")
- Click Save
- Click on back button in browser, which takes you back to the main Machine Control page, so I have to go through all 8 of these steps again for each of the files I want to download.

Is there a chance you could simplify this in a future release?  One click for all files might not be possible, but if I could just see a "directory" of all the files and then right click&Save As for each file (with no name/extension changing and no going back to Machine Control), that would be much faster.

Thanks again.

iDeveloUK, 2017-03-16:
i presume this could be done with JSzip or zip.js?

chrishamm, 2017-04-19:
I'll wait for dc42's DuetWiFiServer rewrite before I address this. It makes no sense to implement that as long as the Duet WiFi cannot handle concurrent connections.

chrishamm, 2017-07-10:
Implemented as part of DWC 1.17.

closed 2017-07-10 by chrishamm