Duet3D/DuetWebControl #35: Comma in .gcode filename causes error

MortarArt, 2017-04-14:
[image]

**Issue description:**

.gcode file list parsing does not complete. (See Stars before names in attachment)

**Steps to replicate:**

Upload .gcode file with ' symbol in filename.

**Variations to issue:**

Navigate to System Editor, wait a minute, and return to .gcode list, and all files, except the one with the bad filename are parsed.

Not sure how to delete the file, as the relevant button doesn't appear until parsing is complete.

MortarArt, 2017-04-14:
Hope I'm in the right place for this.

chrishamm, 2017-04-19:
Yep, that's the right place. To delete it I suggest you take out the SD card once and delete it using a PC.

I'll change DWC to check filenames for illegal characters before they can be uploaded.

closed 2017-04-20 by chrishamm