Duet3D/DuetWebControl #38: Add up and down arrows to the "Send G-code" box.

timmit99, 2017-05-08:
It would be really helpful to add an up-down arrow functionality to select previous commands that have been manually sent to the machine.

chrishamm, 2017-05-19:
The problem is that certain browsers (Chrome/Chromium in particular) don't save each input in the auto-complete suggestions. Hence I've been wondering if it makes sense to move the browser-based auto-complete list entirely to the default G-code list that pops up when the button next to the input is pressed.

lederer-usc, 2017-05-27:
unsure how to answer the question chrishamm started above as i don't really recall how it works currently , but she reading the original post , i can definitely send that notion that a way to reissue commands written on the gcode input line would be very helpful .  on panelDue , we can simply press the up arrow , as mentioned ,which is very convenient .

chrishamm, 2017-07-12:
AFAIR DWC 1.16 already allows G-code entries to be auto-completed, so when you use the up/down arrow keys on the G-code inputs there, it should suggest the last G-codes you entered. This may not work well in Chrome/Chromium though, because there has been a bug which causes G-codes to be saved only once after the web interface is loaded. I don't know if this has been fixed yet.

closed 2017-07-12 by chrishamm