Duet3D/DuetWebControl #41: Not updating

Kulitorum, 2017-05-25:
When I upload the DuetWebControl-1.16.zip file to my Duet wifi, it does not update, but stays at version 1.15c. Am I doing it wrong, or is there something else I need to do?

To upload it, I press "upload file" inside the webpage. A popup window shows unzipping each file and then the duet seems to reboot. But after, it's still at 1.15c

chrishamm, 2017-05-25:
Don't forget to update RRF and DWS to 1.19. However be aware that dc42's and my DWS+RRF versions may not be mixed as they are currently not compatible. See this guide for further information: https://duet3d.com/wiki/DuetWiFiFirmware_1.19_alpha

closed 2017-05-25 by chrishamm

Kulitorum, 2017-05-25:
So how can I get the source for 1.15c? - I need to make some changes.