Duet3D/DuetWebControl #43: Add date uploaded and allow sorting of G-Code file columns.

RichWP, 2017-05-29:
Hello,

It would be great I could see the date when a G-Code file was uploaded - it would also be great to be able to sort by this date and the other columns in 'G-Code Files' tab.

Thanks.

yonkiman, 2017-06-12:
I'll second that suggestion.  Operation would be easier (and I'd be less likely to accidentally print the wrong file) if there was a sort mode where the latest file uploaded was always at the top of the list.

chrishamm, 2017-07-12:
By hovering the mouse on G-code files it is already possible to see when the file was last modified - adding a new timestamp isn't supported by FatFs, so this is no option.

Columns will be sortable in DWC 1.17.

closed 2017-07-12 by chrishamm

yonkiman, 2017-07-14:
Do you mean you can't change the existing "last modified" timestamp, or you can't create a new "date imported" timestamp field?  If you can't change the "last modified" date, how about just a text file with filenames and date imported data?

And will you be adding "last modified" to the columns so we'll be able to sort by that date?  That's better than nothing and would achieve 90% of what I (and probably @RichWP) are looking for: newest files on top.

Alex9779, 2017-07-16:
I also missed the change date but I recently found that hovering shows the date.
But I also vote for adding it as a column.
What I really don't need really is the "Generated by" column, maybe you could just switch those informations? Hovering will display the generating app and the column can be used for displaying the modified date?