Duet3D/DuetWebControl #49: 3D grid view Z value direction

Alex9779, 2017-07-12:
I just realized that the whole display of the bed is flipped at least for my setup.
Is that by intention?
I mean when I look at the 3D representation I expect to see how my bed is but in reality it is just the other way. What is too hight in the picture is actually too low and vice versa...

chrishamm, 2017-07-12:
I cannot confirm this, it is all displayed well on my machine.

- What firmware version are you using and does this problem persist with DWC 1.17-RC6?
- How would you expect the visualization to display your probe points? The values reported by the firmware are deviations from the true level plane.
- Can you provide a screenshot?

Alex9779, 2017-07-12:
WebIF is 1.16...
Hmmm I am not really sure, I thought too it was correct but last time I accidentially ran the compensation with something left on the plate. And where that residue of the last skirt was the grid showed spikes in -Z direction...
I think I am going to check that again with the new version...
Sorry...

closed 2017-07-12 by Alex9779