Duet3D/DuetWebControl #50: G-Code console log messages weird

Alex9779, 2017-07-26:
Hi,
I have no idea when this started, currently I am on version 1.17+2.
The issue is that the console always repeast previous log messages and does not show message for commands I enter.
This works for the first ocmmand, but later when I send other ocmmands I always get the result of earlier commands printed...
```
10:01:56 M307 H1
Auto tuning heater 2 using target temperature 240.0C and PWM 1.00 - do not leave printer unattended
Auto tune phase 1, heater on
Auto tune phase 2, heater off
Auto tune phase 3, peak temperature was 245.1
09:59:26 Auto tuning heater 2 using target temperature 240.0C and PWM 1.00 - do not leave printer unattended
Auto tune phase 1, heater on
Auto tune phase 2, heater off
Auto tune phase 3, peak temperature was 245.1
09:59:20 M303 H0 S90
Auto tuning heater 2 using target temperature 240.0C and PWM 1.00 - do not leave printer unattended
Auto tune phase 1, heater on
Auto tune phase 2, heater off
Auto tune phase 3, peak temperature was 245.1
09:58:18 M307 H1
Auto tuning heater 2 using target temperature 240.0C and PWM 1.00 - do not leave printer unattended
Auto tune phase 1, heater on
Auto tune phase 2, heater off
Auto tune phase 3, peak temperature was 245.1
09:58:17Message Log cleared!
```
In the above log I wanted to read PID paramters but I can't...

chrishamm, 2017-07-26:
This is most likely a firmware issue. What board and firmware version are you using?

Alex9779, 2017-07-26:
Duet WiFi 1.0 + DueX2
1.19beta11 (2017-07-25)
1.19beta10
1.17+2

chrishamm, 2017-08-09:
Can you try that with another browser and see if the problem persists? Which one were you using when it happened? It looks like your browser is aggressively caching the G-code response, although I think we added some counter-measures to avoid that.

Please copy the output of M122 as well when it occurs next time.

closed 2017-09-08 by chrishamm

darookee, 2017-11-21:
I have the same problem, tested using Chrome, Firefox and Internet Explorer (Edge). It does not look like a browser-cache problem to me, because it also happens with 'Disable cache' checked and the Chrome developer tools open.

I tried it just now, sending M122 as the first G-Code, M81 after that, getting the M122 response again. In the network-tab it shows that `rr_reply` contains exactly what is displayed in the G-Code console log.

# DuetWifi info
Hardware: Duet WiFi 1.0
Firmware: 1.20beta8+1 (2017-11-18)
WifiServer: 1.20b9
DWC: 1.19.3

# M122

```
=== Diagnostics ===
Used output buffers: 4 of 32 (14 max)
=== Platform ===
RepRapFirmware for Duet WiFi version 1.20beta8+1 running on Duet WiFi 1.0
Board ID: 08DAM-999TL-MQ4S4-6JKD6-3SD6S-TNGHY
Static ram used: 15488
Dynamic ram used: 99264
Recycled dynamic ram: 4032
Stack ram used: 1368 current, 9248 maximum
Never used ram: 3040
Last reset 14:17:31 ago, cause: reset button or watchdog
Last software reset reason: User, spinning module GCodes, available RAM 7352 bytes (slot 3)
Software reset code 0x0003, HFSR 0x00000000, CFSR 0x00000000, ICSR 0x00400000, BFAR 0xe000ed38, SP 0xffffffff
Error status: 10
Free file entries: 10
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest block write time: 16.0ms
MCU temperature: min 23.7, current 25.7, max 30.1
Supply voltage: min 1.3, current 1.7, max 12.6, under voltage events: 1, over voltage events: 0
Driver 0: standstill
Driver 1: standstill
Driver 2: standstill
Driver 3: standstill
Driver 4: standstill
Date/time: 2017-11-21 11:28:59
Cache data hit count 4294967295
Slowest main loop (seconds): 0.061329; fastest: 0.000033
=== Move ===
MaxReps: 5, StepErrors: 221, FreeDm: 240, MinFreeDm 122, MaxWait: 4291329298ms, Underruns: 0, 0
Scheduled moves: 0, completed moves: 0
Bed compensation in use: mesh
Bed probe heights: 0.000 0.000 0.000 0.000 0.000
=== Heat ===
Bed heater = 0, chamber heater = -1
Heater 0 is on, I-accum = 0.0
=== GCodes ===
Segments left: 0
Stack records: 3 allocated, 0 in use
Movement lock held by null
http is idle in state(s) 0
telnet is idle in state(s) 0
file is idle in state(s) 0
serial is idle in state(s) 0
aux is idle in state(s) 0
daemon is idle in state(s) 0
queue is idle in state(s) 0
autopause is idle in state(s) 0
Code queue is empty.
Network state is running
WiFi module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
WiFi firmware version 1.20b9
WiFi MAC address 60:01:94:0c:51:38
WiFi Vcc 3.46, reset reason Turned on by main processor
WiFi flash size 4194304, free heap 28928
WiFi IP address 192.168.0.192
WiFi signal strength -44dBm, reconnections 0, sleep mode modem
HTTP sessions: 2 of 8
Socket states:  2 0 0 0 0 0 0 0
Responder states: HTTP(1) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0)

Warning: VIN under-voltage event (7.3V)
```

darookee, 2017-11-21:
As mentioned [here](https://www.duet3d.com/forum/thread.php?pid=30734#p30734) it is due to another client not GETting '/rr_disconnect'. Could be another Browser or any other client connecting. This fixed it for me, too.