Duet3D/DuetWebControl #60: [Request] Custom Macro Buttons

tjschweizer, 2017-10-26:
I'm poking through the code trying to add it myself, but I think it'd be nice to have some "hot buttons" like right below the heater controls to run macros. Just bind a button to a macro to be able to run it from any screen.

pkmcnc, 2017-10-27:
At least two macro buttons "Preheat" and "Cool down" below the heater controls would be very nice.

chrishamm, 2017-11-12:
Why is it not sufficient to create two macros and invoke them on the Machine Status page?

In principle adding new buttons to DWC isn't difficult - you'll only need some HTML like
<a href="#" data-gcode="M98 Pcooldown">Run cooldown macro</a>

tjschweizer, 2017-11-14:
That is definitely sufficient, running the macro from the Machine Control page.

I run my printer at work, and remote into it from home to check the status once in a while. I have some LED lights hooked up to a fan output. In order to turn the lights on, I have to turn the fan on using the slider (difficult on a phone using TeamViewer) or by switching to Machine Control and running a macro, then go back to the Print Status page to view the webcam feed.

Really I am just being lazy, but I didn't know if this was something others had interest in as well.

So the HTML to add the button is pretty simple, but linking it to a macro is a bit more involved. I had some parts to print and haven't had a chance to test much code yet.

chrishamm, 2022-01-29:
There is now the BtnCmd plugin for DWC which lets you customize a lot, so I am closing this request.

closed 2022-01-29 by chrishamm