Duet3D/DuetWebControl #61: Move API calls to Duet Wiki

Kriechi, 2017-10-31:
Thanks for the great project!

I was playing around with the Duet RepRap firmware API calls in the last days, and always had to read the actual C++ implementation to figure out how the HTTP API is supposed to work.
Now I finally stumbled across the documentation I was looking for the whole time in the README here: https://github.com/chrishamm/DuetWebControl/blob/742fa9138ff4fb32be068b1172535678c1d85fa5/README.md#list-of-http-requests

IMHO this section should be moved to the Duet Wiki: https://duet3d.com/wiki/Duet_Wiki, since the code is actually part of RRF and not DWC. DWC is "just a client" using it.

Thanks for documenting it! I was about to write a wiki page myself, but now I think @dc42 could just cut & paste it.

chrishamm, 2018-06-26:
Hmm, I see no point why we should document it twice. Basically my readme just documents which calls the web interface makes and the HTTP interface can be viewed either from RRF's or DWC's side. So if you want to clone the documentation on the Duet3D wiki please go ahead, but I won't maintain that.

closed 2018-06-26 by chrishamm

Kriechi, 2018-06-27:
Just to clarify, I didn't mean to duplicate the information. 
As the title of this issue suggests, I proposed "moving" the API documentation. Currently this repo documents something it has no control over because the actual implementation of the API is in a different project/repository/software.

Kriechi, 2019-10-10:
@pctechjon the most important bits are probably here: https://github.com/dc42/RepRapFirmware/blob/51bf2c4ea69464882357623d45fa7d238975e269/src/Duet/Webserver.cpp#L28-L83