Duet3D/DuetWebControl #62: [Feature] Automatic reconnect

Kriechi, 2017-11-09:
It would be nice if DWC would try to automatically reconnect (after a timeout).

Every time I put my notebook to sleep, and wake it up again, DWC complains about loosing the connection. Having it reconnect automatically would give a more streamlined experience.
Doesn't have to be on-by-default, an options / checkbox would be nice though!

chrishamm, 2017-11-12:
Actually DWC does that, that's what the AJAX retries are for - by default DWC only attempts to reconnect once, but you can increase this value on the Settings page (Maximum number of AJAX retries). You may need to increase the timeout value too.

I'll check if I can enhance DWC further to deal with suspend events, but I can't promise anything.

Kriechi, 2017-11-12:
Yes - I already played around with those.
But considering I start a print & close my notebook, at some point I will run out of "retries".
I guess MacBooks do all sorts of "non-standby" things even with a closed lid!?

Not sure if this is still valid today, or how hacky it is, but maybe this could do the trick: https://blog.alexmaccaw.com/javascript-wake-event

chrishamm, 2017-11-12:
I think that will come in handy - thanks!

chrishamm, 2018-07-31:
Implemented in 1.22-b1.

closed 2018-07-31 by chrishamm