Duet3D/DuetWebControl #63: Display heater names instead of "Heater 1"

timmit99, 2017-11-10:
Using the M305 command in RepRapFirmware i have named my heaters for the left and right extruder. I would like it if the Duet Web Interface would display those names to make it more descriptive.

My current command is `M305 P2 S"Left Extruder" X205 H0 L0` for an FDM extruder using a PT100 on channel 205. It would be nice if the web interface could pull the heater names into the web interface.

Gwhittredge, 2018-03-16:
With the tool names showing up this the web interface this would be very useful for any custom machines that use build chamber heaters and multiple heaters per tool.

timmit99, 2018-06-07:
Any chance this could be implemented?

chrishamm, 2018-06-30:
Has been implemented, please try out DWC 1.21.2-b2 and my latest firmware version.

closed 2018-06-30 by chrishamm

timmit99, 2018-07-01:
So to test this i would need the DWC from [HERE](https://github.com/dc42/RepRapFirmware/releases/download/2.01beta1/DuetWebControl-1.21.2-b1.zip) and YOUR branch of reprap firmware from [HERE](https://github.com/chrishamm/RepRapFirmware/blob/dev/Release/Duet-WiFi/DuetWiFiFirmware-1.20.1.bin)? Or can i use DC42's branch of RRF?

chrishamm, 2018-07-01:
No, you will need these two files:

https://www.dropbox.com/s/kbjp5pp3wowwhot/Duet2CombinedFirmware.bin?dl=0
https://github.com/chrishamm/DuetWebControl/blob/dev/DuetWebControl-1.21.2-b2.zip