Duet3D/DuetWebControl #66: [Feature] Print History

lesaux, 2018-01-27:
A tab with print history would be nice. Maybe its possible to keep a print history on sdcard?

nuker1991, 2018-01-27:
Also with statistics like meters/hours printed is nice :)

chrishamm, 2019-02-11:
This would have to be implemented in the firmware first - the web interface can only display what the firmware provides. Any other solution would be rather dirty. Expanding the rr_fileinfo response could be a solution.

closed 2019-02-11 by chrishamm

JohnOCFII, 2019-02-12:
Actually, the most important data to display is already captured when the event log is being created:  Just having a tab to display this data in a nicely formatted way would be great.  Here are two entries that could be used, for example:
`2019-01-31 18:28:16 Started printing file Calibration/3DBenchy_0.2mm_PLA_new_nozzle_RC.gcode
2019-01-31 20:00:28 Finished printing file Calibration/3DBenchy_0.2mm_PLA_new_nozzle_RC.gcode, print time was 1h 32m
`