Duet3D/DuetWebControl #77: [feature] Select time format

wilriker, 2018-05-03:
Currently times are displayed in 12h AM/PM format. It would be nice to have an option to change this to 24h format.

chrishamm, 2018-05-03:
If you are talking about the end times then the formatting depends on your browser configuration. DWC uses the locale as configured in your browser so I suggest you change that and/or your system's formatting settings.

closed 2018-05-03 by chrishamm

wilriker, 2018-05-04:
I talk basically about all times displayed in DWC, e.g. Last Modified columns in all file tables but also about end times.

My systems are configured to use `en_DK.UTF-8` which is English language but European/ISO date and number formats. It works everywhere except with DWC. I also changed all settings in Vivaldi and Firefox to be only German but still I get all times in DWC formatted as AM/PM.

I tried running `new Date().toLocaleString()` in the console of the browser and interestingly this gives me exactly the format that I see in DWC. So here this might be an issue with my browser(s).

Still I think it would be nice to have an option to force date and time format to a specific value.