Duet3D/DuetWebControl #80: G-Code Console issue in 1.21.2-b1

gtjoseph, 2018-06-26:
If you type a G code previously executed, you get a drop down of the previous entries.  Nice feature!
Let's say I just want to execute G91.   I type G91 in the entry field and press enter.  Instead of executing the command, the dropdown appears and I have to hit enter again to get it to accept just the G91.  Same for any command.  If you've run it before, you have to now hit enter twice even to execute only what you typed.

Since G91 is also executed as part of the Machine Control buttons, I also get in the dropdown, all the multiple command sequences from those buttons.  If I hit enter on one of those, I get the html from the dropdown.  :)

`M120<br/>G91<br/>G1 S2 X100 F6000<br/>M121`

I can clear those out in the Default G-Code List Items but they come back of course.

So...  I think adding the commands from the control buttons is a bug but I also think there needs to be an option to control the enter key behavior.  Maybe you need to hit the down arrow key to get the matching list.  

I'll throw in a feature request as long as you're there... :)   Double clicking on a log entry should automatically rerun the command.

chrishamm, 2018-06-26:
The second issue is known and has been fixed in b2 which I have not released yet. I'll check out the first request once again. If there is a simple fix I will implement it.

gtjoseph, 2018-06-26:
Cool, THANKS!

chrishamm, 2018-06-26:
I just finished a work-around which will be included in my next beta.

closed 2018-06-26 by chrishamm

gtjoseph, 2018-06-27:
Thanks again.  I just had a close call while getting my Z max endstops configured.  I typed G1 Z30 and hit enter twice before I realized that I had typed G1 Z300 earlier and that was what was sent.  I had to lunge for the mains power button. :(