Duet3D/DuetWebControl #81: Still having G-Code Console issue in 1.21.2-b2

gtjoseph, 2018-07-02:
Still having the issue where the wrong entry is picked out of history.
M1 Z120<enter> results in the Z axis moving to 120.
M1 Z12<enter> results in nothing being sent.
Pressing enter again results in M1 Z120 being sent again instead of M1 Z12

It's just not usable this way. If I type something and hit enter, exactly what I typed should be sent.  If I want to pick something out of history, then I should use the arrow keys to select the proper entry and hit enter.

chrishamm, 2018-07-02:
Then press ESC to close the auto-completion for now. That does work.

gtjoseph, 2018-07-03:
I wasn't sure if there was supposed to be a fix in b2 or not, sorry.

chrishamm, 2018-07-24:
Alright, I'll fix it in the next beta.

closed 2018-07-24 by chrishamm