Duet3D/DuetWebControl #83: Touch optimized Template

JoduWei, 2018-07-11:
I managed to install the webinterface locally to a raspberry pi/pc. Its really easy. It would be really nice to have a touch optimizied version for smaller displays. I mean a full screen condensed interface without scrolling. Maybe a second template file.
I would have to read in everything, since the template is not splitted into modules.

chrishamm, 2018-07-11:
More customization will be possible in the future but I don't think I will be able to provide that as part of Duet Web Control. It is more likely to expect this as a part of DWC's successor which is still being planned.

JoduWei, 2018-07-11:
"DWC's successor" ?

chrishamm, 2018-07-11:
Yes

JoduWei, 2018-07-11:
Great, tell me, if you need support in form of coding.

Patrik-svobodik, 2018-12-23:
I think it is safe to close this, the new DWC2 is really well optimized for touch devices

chrishamm, 2018-12-23:
I will close this when DWC2 is out.

chrishamm, 2018-12-25:
Please give DWC2 a try, it is better optimized for touch screens.

closed 2018-12-25 by chrishamm