Duet3D/DuetWebControl #84: COREXY error in interface.js causes incorrect moves

rodneykorn, 2018-07-17:
The change on line 1307 in  interface.js 
		if (settings.allowUnhomedMoves && geometry != "delta" && geometry != "scara") {
is adding the S2 on COREXY systems (Promega) which results in a diagonal move when using the DWC jog buttons.
I tested version 
1.21.1, which did not have the issue.
1.21.2.Beta2 did have the issue
Loading 1.21.1 and executing the commands

M120
G91
G1 S2 X100 F6000
M121

results in a diagonal move 
Using the current beta firmware (2.01beta2)

rodneykorn, 2018-07-20:
Another person reported that this also is affecting the Macros as will as the jog buttons.  I have not researched that part of the issue but assume it is probabaly the same issue.

rodneykorn, 2018-07-20:
Note the bug is in the DEV branch Commit 2f4134f2

chrishamm, 2018-07-27:
A fix for this is on the way. In the meantime I recommend turning off the option "Allow moves if axes are not homed" on the Settings page.

Please be aware that 1.21.2-b2 is a beta version and not intended for production use; however it will not take long until I release the next DWC version.

chrishamm, 2018-07-31:
Fixed in 1.22-b1.

closed 2018-07-31 by chrishamm