Duet3D/DuetWebControl #85: [Feature Request] More Filament managemnt options

brunofporto, 2018-07-18:
For now I created a folder with PLA, ABS and PETG materials load and unload macros.

At my filament list I have several PLAs, PETGs, ABS from different brands and I am just calling those "external" macros so it is easy to tweak something.

Would be nice to have the option to control them as individual spools not "filaments"!

And add informations like material (that would automatically use the correct load and unload macros for that material) and manufacturer. This will make it simple in the future when you implement spool management to the system :D

Stephan3, 2019-03-02:
@chrishamm +1

why not store it in jsonformat insde the filament settings? Then count out/in length each status_1 poll as E moves.

chrishamm, 2019-03-02:
DWC should only act as a visualiser for the data it receives from the controller, that's also the reason I removed the "Override fan value" option from DWC2. As long as the spool support hasn't been improved in the firmware, I won't add new facilities to the web interface.

Stephan3, 2019-03-02:
okay understood, clear statement. Thanks. Once its there i will emulate this with klippers abilitys.
Thanks