Duet3D/DuetWebControl #95: [DWC2] Unable to get status data

Patrik-svobodik, 2018-12-22:
Hi, I just compiled DWC2 and it looks stunning, but there is a problem with getting status info, it complains about some timer and some things that disappear quickly, I can home the printer, but it never knows if it is already homed, the same with sending gcode

chrishamm, 2018-12-22:
What board, printer and browser are you using? If you decide to try it out, it's probably a good idea to run it locally via 'npm run dev'. This makes debugging easier because the stack traces in the JS console are more verbose compared to the on-board variant.

Patrik-svobodik, 2018-12-22:
I run it locally on my machine with npm run serve, it fails to connect most of the time but when it does id does not receive status from the board
[image]

Patrik-svobodik, 2018-12-22:
[image]

Patrik-svobodik, 2018-12-22:
BTW I think the DWC should detect if you have a probe and if you do then not show the option to do G32 bed leveling

Patrik-svobodik, 2018-12-22:
I am using a Duet WiFi on RRF 2.02RC6 (the newest)

Patrik-svobodik, 2018-12-22:
and one other question, I can just build this and put it on the SD card?

chrishamm, 2018-12-22:
Thanks, will check it out. What browser are you using?

https://ci6.googleusercontent.com/proxy/iz0NxN0sTKLI4YpoULRZWk7bA_a-ylRoyqvbjLYKsEOAwpw5GjeNtdqX1v1lR-yvI14kLiWh82sZMuUDQe8APFLYtraeFHxNxLjSbO6lLVY7lEfLyi9vr8-qdwgFjsFIrxuwRywhWYNW0LiWeFEQybar9A=s0-d-e1-ft#https://github.com/notifications/beacon/AFpS2Qh_nRXQ6T2DIOlDhJLg5Jo1TkHxks5u7i3zgaJpZM4ZfbTx.gif>

chrishamm, 2018-12-22:
You can build it and upload one of the zip files that are generated in
dist. But I recommend you wait for dc42s next RC first because it improves
support for the new web ui.

wrote:
https://ci6.googleusercontent.com/proxy/iz0NxN0sTKLI4YpoULRZWk7bA_a-ylRoyqvbjLYKsEOAwpw5GjeNtdqX1v1lR-yvI14kLiWh82sZMuUDQe8APFLYtraeFHxNxLjSbO6lLVY7lEfLyi9vr8-qdwgFjsFIrxuwRywhWYNW0LiWeFEQybar9A=s0-d-e1-ft#https://github.com/notifications/beacon/AFpS2Qh_nRXQ6T2DIOlDhJLg5Jo1TkHxks5u7i3zgaJpZM4ZfbTx.gif>

Patrik-svobodik, 2018-12-22:
I use safari 12.0.1 (14606.2.104.1.1)

Patrik-svobodik, 2018-12-22:
just tried deploying it to my duet, it still does not work, although the connecting is almost instant and reliable, but still complains about this:
[image]

Patrik-svobodik, 2018-12-22:
the same happens on connect/disconnect

Patrik-svobodik, 2018-12-22:
and suddenly it cannot connect anymore

chrishamm, 2018-12-22:
Feel free to try a 'git pull' and test it again, it's working now. Bear in mind that this version is NOT official yet and still in active development.

closed 2018-12-22 by chrishamm

Patrik-svobodik, 2018-12-22:
It works now, thanks a lot

Patrik-svobodik, 2018-12-22:
I see a lot more issues, should I open a new issue?

Patrik-svobodik, 2018-12-22:
[image]

[Záznam obrazovky 2018-12-22 v 19.42.16.zip](https://github.com/chrishamm/DuetWebControl/files/2705460/Zaznam.obrazovky.2018-12-22.v.19.42.16.zip)

Patrik-svobodik, 2018-12-22:
for some reason the class="grey lighten-4" is added to the <tr data-v-5e8d53b0=""> element when the tool is selected and starts heating, can you do anything about it?

chrishamm, 2018-12-22:
> for some reason the class="grey lighten-4" is added to the element when

As I said the 'next' branch is NOT for production because I am still
working on a lot of details and it is NOT finished yet. Once I release a
first version and you notice problems, feel free to open new issues.

Patrik-svobodik, 2018-12-22:
I know it is not ready, I am jist pointing out issues

chrishamm, 2018-12-22:
Sure, thanks. Here my current TODO list:

- file names => inline
- document title => currently selected machine (+ print progress)
- add piecon support again
- utils/merge performance evaluation
- sd card mount: remove error message
- test on-board cache

Let me know if you find anything else!

Patrik-svobodik, 2018-12-23:
You should invert the default sorting so that it is identical to the old
DWC, or better, make ut so that it remembers what you set, when loading the
GCODEs the whole interface (and sometimes my whole mac) hang and will be
not responding untill it is fully loaded

Patrik-svobodik, 2018-12-23:
And add option to not show every gcode that has been run from the UI, it is annoying (as the copyright on the bottom is, but you said it will be moved)

Patrik-svobodik, 2018-12-23:
and one more thing I find annoying- the button "Start another" is missing

Patrik-svobodik, 2018-12-24:
BTW is the language picker present as of now? I am working on a Czech translation but I am unable to test it

chrishamm, 2018-12-24:
Thanks for reporting the issue with the "Start another" - I'll try to reproduce and to fix it.
i18n is prepared but I haven't found more time to prepare the translations for everything but the start page. It's on my TODO list.

Patrik-svobodik, 2018-12-24:
I am already working on the czech translation, I just duplicated the en.js
file and added it to the index.js, will it change? If yes I will wait and
translate it later

Patrik-svobodik, 2018-12-24:
another problem, when I send M999 in the console it hangs because it is expecting a response, Merry Christmas BTW

Patrik-svobodik, 2018-12-25:
another thing - I am unable to upload the doc files

Patrik-svobodik, 2018-12-25:
dwc files

Patrik-svobodik, 2018-12-25:
and if I upload the firmware files it does not prompt me to update

chrishamm, 2018-12-25:
Use the Files/System page

https://ci3.googleusercontent.com/proxy/Q_jtNw193B1dpMrZLimHy77sB36HmXkrmWtGYW1GUvp5rqjuyz_pzs47kUgxEXfY3XlYgcPABZCPKFZz6Gdl71xXKwF0Yiislvsiqp3WMmt3ZfWgGDpBtOoSfRU1hfdEPHEg1_pHCmFBsv1DiidppIQIfQ=s0-d-e1-ft#https://github.com/notifications/beacon/AFpS2VUrNcq2U2ykDUCT-0os8g6fiDFaks5u8p79gaJpZM4ZfbTx.gif>

Patrik-svobodik, 2018-12-26:
you broke the print dialog again in RC2

chrishamm, 2018-12-26:
> this one

Works perfectly fine on my setup. What is wrong on yours?

Patrik-svobodik, 2018-12-27:
Don’t get me wrong, it works, but it should have ‘overflow:’ hidden; set but it does not

Patrik-svobodik, 2018-12-27:
And one more thing: the G32 bed leveling should be gone if a DWC detects a probe on a cartesian or coreXY machine