Duet3D/DuetWebControl #96: [DWC2] No option to select the machine to connect to on mobile

Patrik-svobodik, 2018-12-22:
Hi, when I try to connect to my npm dev server from my iPhone it works "fine" (the bug from the other thread is still present) but for some reason when I click connect it doesn't let me choose the ip to connect to, it then just loops forever trying to connect

chrishamm, 2018-12-22:
Not a bug. DWC tries to connect to the remote ip if it does not run on
localhost so I won't address this. Multi machine support will be eventually
added but that will happen later when I have more time.

So what browser are you using on your computer?

> Hi, when I try to connect to my npm dev server from my iPhone it works
"fine" (the bug from the other thread is still present) but for some reason
when I click connect it doesn't let me choose the ip to connect to, it then
just loops forever trying to connect
https://ci4.googleusercontent.com/proxy/pXB1pXPp284-GUsdEc_Ezqw0s4h12BVJuSntaIELerQNzEQ85J-2aqIRtFMBEydJP2lGFoamNAv7VWdb7XEELbfqg2E-myphFiW2VHp_KYQAW3AscWjNS6ygqSzZJTc0FcQKH970f1oKfI0BH5UAF0LKGQ=s0-d-e1-ft#https://github.com/notifications/beacon/AFpS2eWRW7rTHNanw1XkBtE8FY8y0M_0ks5u7i9xgaJpZM4ZfdyQ.gif>

Patrik-svobodik, 2018-12-22:
I use safari on both

chrishamm, 2018-12-22:
OK, thanks, I could confirm there was a bug in the reconnect logic which should be fixed now.
Eventually I will remove the connect button like I did in DWC1 but for now it can be handy for testing.

closed 2018-12-22 by chrishamm