Duet3D/DuetWebControl #103: [Feature Request] Macros Uploading

johnkray, 2019-01-08:
I work for a small company that is releasing a duet based 3d printing soon and we are hoping it might be possible to add a better way of updating macro files as we use quite a few of these.

I'm wondering if it is possible to add the ability to upload macro files contained in a zip along with the system, firmware, and DWC files. I'm also wondering if the macro file structure can be preserved in this process. Essentially, we would like to push one zip file that contains all these things to our customers so they do not have to manually upload macros separately if the macros have been changed.

chrishamm, 2019-01-08:
For this purpose I am planning on adding a new (Auto-)Update page. The code for this is already mostly finished in DWC2 but I need to find some more time to actually add it. In the meantime you have to ship two update packages for /sys+/www and /macros. Firmware and web interface updates can be already combined in a single ZIP.

johnkray, 2019-01-08:
That sounds great! 

You said auto update, does that mean it will be able to check an online repository for new versions and prompt the user to update if there is one?

chrishamm, 2019-01-08:
Yes, that is the idea. GitHub provides an API for this purpose but DWC does not use it yet.

johnkray, 2019-01-08:
Fantastic! I'm looking forward to the implementation.

chrishamm, 2020-01-13:
If you want to provide your own update packages, it should be possible to install them by zipping up the entire SD directories. Just make sure the files are contained in the actual directories (e.g. /sys for system files, /www for HTML files, /macros for macro files etc.). You can then upload such a package on the System page.

I tried to change the config tool to download and integrate the latest DWC/RRF versions automatically but it's only partially possible because of GitHub's inconsistent CORS headers. Hence I cannot make DWC2 download GitHub assets directly - and unfortunately GitHub isn't willing to address this.

closed 2020-01-13 by chrishamm