Duet3D/DuetWebControl #105: [Feature Request] Allow numeric input on all speed sliders

toskium, 2019-01-16:
Currently fan and print speed can only be set using a fiddly slider. It works well from a desktop computer utilizing a proper pointing device but as soon as one uses a small screen on a mobile phone the speed sliders become unusable.

It would be great if the speed sliders could optionally be replaced by input boxes to set a percentage and apply afterwards.

mydevpeeps, 2020-08-25:
Suggestion: 
Rather than optionally replacing the slider, or having an input box .. add a drop-down box with an increment that is defined in the config section (such as 5%, 10% etc) which when clicked would have the same effect as moving the slider. This would then in turn actually move the slider to the chosen item.  Have a configuration item that would enable/disable this drop-down or make it automatically show up on mobile/tablet.

Example:
Fan Slider is at 50%. User clicks the drop down and sets it to 75%. UI sends Request URL: http://ip.of.duet.board/rr_gcode?gcode=M106%20S0.75 and then adjusts the slider display to match 75%.

chrishamm, 2022-01-29:
This feature will be part of the next version, although it must be enabled on the Settings -> General page first.

chrishamm, 2022-02-10:
Now implemented in 3.4-rc1.

closed 2022-02-10 by chrishamm