Duet3D/DuetWebControl #107: KlipperWebControl - possible?

hevilp, 2019-01-21:
Hello,

I love the DWC on my DuetWifi.
For another printer I need to use Klipper, is it possible to port it to klipper firmware?

https://github.com/KevinOConnor/klipper

The author of Klipper is very active and I think he would stay at your side to get it work.

Thanks in advance

BlackStump, 2019-01-24:
I too have a Duet2 Ethernet and think the DWC is great and it would be awesome to be able to use the DWC on the other board/printer with Klipper.

Thanks in advance

Stephan3, 2019-02-22:
Dont chat, just do it.

https://github.com/Stephan3/dwc2-for-klipper

chrishamm, 2019-02-26:
Well done, @Stephan3! I'm afraid I won't officially provide support for Klipper because I don't use it, but it's nice to see you managed to add support for it.

I'm about to push RC4 to GitHub and it will include some changes to the object model, in particular
- job.file is renamed to job.fileName
- job.filamentNeeded is renamed to job.filament
- most internal model values must accept 'null' values in preference of 'undefined'
Just FYI

closed 2019-02-26 by chrishamm

Stephan3, 2019-02-26:
@chrishamm thanks for the Feedback. I tried to keep both "world's" untouched to keep it in a maintainable state. What are your thoughts about a plugin system? So one can put a custom js to a subfolder and dwc imports theese if it's there?

Great front-end btw :)

chrishamm, 2019-02-27:
A plugin system is planned but it will take a while longer before I address it. It may make more sense to design that once the Duet 3 is ready.

TBH I thought you had developed a custom connector for DWC2 that uses the Klipper HTTP API (if there is any) which is why I pointed out the modifications to the object model in my last comment. Adding support for new protocols in DWC2 is quite straight-forward by adding your own JS connector in src/store/machine/connector. Once the whole thing is fully stable, I'll document this part as well.

Anyway, since the current HTTP API won't be changed again, your Python-based translator should stay functional too.

Stephan3, 2019-02-27:
Thanks for the answer sounds good. Today there is no Api in klipper. I thought about that too. But then it needs a standard protocol. Basicaly it is allready there. I see more sense in translating klippers data into a format which dwc is allready using. 

People are asking me for pluginfunctionality like loadcell for spoolweight. Filament length used from spool etc. I didnt mange the filament thing currently present in dwc2.