Duet3D/DuetWebControl #115: File/folder selection behavior changed from RC4

JohnOCFII, 2019-03-01:
Greetings,

Testing RC-5, I noticed the file or folder selection process is different. Is that intentional?  First click highlights the file or folder (and fills in the check box), and second click opens the folder or initiates action (such as print).

I'm OK with the new behavior.  Just noting that it is different.

This is in both Safari and Chrome on the Mac.

chrishamm, 2019-03-01:
Yes, it's new and basically the same behaviour as in every other file browser (Explorer, Finder etc). The link in the file name column is gone now because that improves readability when the dark theme is active.

I may even change the click behaviour in the file tables again depending on the input type so touch clicks/double clicks would be interpreted differently. For example when using DWC on a touch screen, a single click could activate an item and a double click could open the context menu. At least that would allow iPhone/iPad users to open the context menu, which is not possible at the moment AFAIK.

closed 2019-03-01 by chrishamm

JohnOCFII, 2019-03-01:
I agree readability in dark mode is much improved.   I just tried this in Safari on my iPad, and I'm not able to click into folders in the G-Codes directory since double-click isn't recognized.

chrishamm, 2019-03-02:
Does a long press open the context menu on your iPad (especially on the Console page)?

JohnOCFII, 2019-03-03:
> Does a long press open the context menu on your iPad (especially on the Console page)?

Not with RC5.

ctilley83, 2019-03-03:
I have to admit, this new behavior is a little wonky. It might make a little better sense if clicking a folder once, didn't select the checkbox.  Maybe limit selecting to the checkbox element itself? There's something unsettling from a UI perspective when you see the checkbox state change while double clicking a file folder.