Duet3D/DuetWebControl #116: Firefox on Android

melbuh, 2019-03-02:
Hi again, 

Scaling on 4k looks nice now in RC5, thank for that!

Unfortunately Firefox on Android is now not supported anymore (Chrome looks fine): 
[image]

From my personal wishlist: Could you add sth do choose the stepsize for the +/- buttons at panel.speedFactor.caption? 5% Jumps are a little rough, and my Wurstfingers are too fat for small steps on the smartphone screen using the slider O:-)

thx in advance! 
Greetings

chrishamm, 2019-03-02:
It looks like the selected language is wrong in your FF mobile - try to revert to the factory defaults and see if that changes anything (I use FF on Android myself and it looks nice there). At present only English is supported since I haven't found time to translate everything to German yet and if DWC attempts to load an unsupported locale, that's what it looks like now.

I'll add support for custom +- steps in the future and allow an option to replace the sliders with numeric inputs, but that will take a while. It's no big deal but I'm still tied up with other stuff.

Can you attempt to reload the page and check if that fixes the input boxes (perhaps even via Ctrl+Shift+R)? I'm afraid I imported some kind of bug from the UI framework DWC2 uses when I updated it, which causes the inputs to malfunction occasionally.

Stephan3, 2019-03-02:
@chrishamm @melbuh 

I noticed the same. As i am on a custom webserver, i gave that one a closer look.
If a dwc2settings.json is existing from a previous version, the json dump lacks the pressence of the key language, therefore the setting isnt present.

If dwc2settings.json is not existent on the host, dwc2 requests a default json which even a rrf duet does not supply. So if my lowlevel oppinion is right i would propose a "if language is null, language = en" in pollcollector?

chrishamm, 2019-03-02:
@Stephan3 This problem has nothing to do with the PollConnector because the connectors only wrap around the responses of RRF in order to make them compatible to the internally used data structures.

The default language key is set to 'en' as in src/store/settings.js:12 and when a JSON object is loaded, non-existent fields are skipped. So my best guess is that there is something wrong in the stored settings which can be resolved by performing a factory reset.

The default value of the language field remains 'en' even when dwc2settings.json does not exist; I just verified that.

Stephan3, 2019-03-02:
yes you are right. My browser was caching "a thing", sorry for bothering.

chrishamm, 2019-03-13:
@melbuh Please comment on my suggestions. If I don't get any more feedback, I'll close this issue.

closed 2019-03-25 by chrishamm