Duet3D/DuetWebControl #122: 2.0.0 Cannot access file list context menu on touch devices

spill50, 2019-03-12:
You can select files in the file list by tapping on them but the is no way to access the context menu on each file on touch devices making it impossible to manage files or start prints from the file list when using a mobile device as there is no right click mouse button.

A couple of possible solutions could be:

- Add touch events to the list items
- Add an “Options” button which triggers the context menu for each list item

chrishamm, 2019-03-12:
Try long-pressing file items. I added touch events to the file lists in RC6 but since I don't have any iPhone/iPad I could not test them on those devices.

spill50, 2019-03-12:
Ah hang on I’m still on RC5, I will update and check again.

spill50, 2019-03-12:
Yes, that works on RC6 thanks. It might be worth just having the single tap to display the menu though. Rather than having single tap dialog to start print and press and hold to get the menu.

On iOS a press and hold on elements causes the browser to go into drag and drop mode for that element which can cause a bit of odd behaviour sometimes.

spill50, 2019-03-12:
This screen capture shows what happens [
https://drive.google.com/file/d/1k32W2d-LRg5BgooemihIStd7mcrzwj1o/view?usp=drivesdk](
https://drive.google.com/file/d/1k32W2d-LRg5BgooemihIStd7mcrzwj1o/view?usp=drivesdk)

chrishamm, 2019-05-07:
Yes, that makes sense. I'll consider that for a future version.

chrishamm, 2020-01-15:
It works on iOS 13.3 so I am closing this issue.

closed 2020-01-15 by chrishamm