Duet3D/DuetWebControl #123: 2.0.0 RC6 Arrows not working for navigation inside file editor

mildw4ve, 2019-03-13:
Using Firefox I often get an issue where when editing any file (mostly I edit config.g etc.) the arrows don't work for moving the typing cursor. Whenever that happens I either have to move the cursor with a mouse or refresh the whole page (the latter makes it work for a little while but the issue comes back).

chrishamm, 2019-03-13:
I'm using Firefox, too, and I haven't had any problems with it - the arrow navigation works well for me. Can you tell me how to reproduce this problem and does it persist when you're in Private Browsing mode?

Fulg, 2019-03-13:
I have the same problem, it seems to be somewhat random. I haven't managed to nail it down. I am using Chrome. Often (maybe 50% of the time?) the arrow keys don't work to position the cursor in the editor.

mildw4ve, 2019-03-13:
I just talked to @Fulg and we both use ublock so its just a guess but perhaps ublock is to blame.

Fulg, 2019-03-13:
I added an exclusion for DWC inside uBlock, I'll report back if it solved the issue.

mildw4ve, 2019-03-16:
I disabled ublock completely for DWC2 and I'm sorry to say it did not help. I'm unable to specify how exactly does this happen, I do the usual stuff, open DWC, play around with configs, run true bed leveling, homing and at this very moment it's broken. I cant use arrows in the editor or in the gcode terminal (both the top and the full one).

mildw4ve, 2019-03-16:
Just found something. 
When I refresh DWC and go to type something in the top GCODE terminal it works, arrows work and all that. Then I go to the "height map" tab and click back on the top GCODE terminal and try to use arrows to move left to right and instead of the typing cursor the whole height map moves left to right. Then I go back to Dashboard and the arrows aren't working there too, nor do they work in the System files editor. Finally at that point when I go back to the height map its often missing, the only way to display it is to click top view but when I try to rotate from there it disappears immediately as if its offset in the Z axis.

chrishamm, 2019-03-26:
Thanks for your observation - it's probably the same component that caused issues in RC5. I downgraded it to the version that I used in RC4 before. Could you try it out? I put up another build here: https://chrishamm.io/dwc2/DuetWebControl.zip

mildw4ve, 2019-03-31:
Just tried it.

1. First in dashboard, arrows work in the top terminal
2. Then in heightmap using arrows in top terminal works but the heightmap moves too, not sure if thats intended
3. Back to dashboard arrows still work
4. Then in System, arrows inside file editor still work

chrishamm, 2019-05-14:
Please check if it is fixed in RC7.

mildw4ve, 2019-05-15:
Sorry to report the issue is still there, same as with RC6, the moment I open the heightmap arrows stop working in top terminal and when I go to edit config.g or any other file arrows don't work there either.

fundix, 2019-06-12:
Solved in RC7?

mildw4ve, 2019-06-12:
Just tried it and it seems the same as in this post of mine https://github.com/chrishamm/DuetWebControl/issues/123#issuecomment-478332199

Arrows don't stop working after I open the heightmap page, but the heightmap moves when I use arrows in the top terminal and the heightmap disappears same as previously.

Not fully fixed but it is better than RC6

chrishamm, 2019-06-18:
OK, thanks. I will disable the arrow control of the heightmap in the next release.

chrishamm, 2019-10-01:
Fixed in v2.0.0.

closed 2019-10-01 by chrishamm