Duet3D/DuetWebControl #124: 2.0.0 RC6 - Broken temperature graph when DWC is left open overnight

Fulg, 2019-03-13:
Leaving DWC up overnight makes the temperature display look quite interesting... :) It seems to show the entire timeline from the first time I connected to DWC _until the current time_.

[image]

My PC usually sleeps during the night (so the connection with the Duet is lost for a while, even though this does not trigger a disconnection message), I don't know if it's related. I can try later disabling overnight sleep.

Patrik-svobodik, 2019-03-13:
I left it while we were on vacation, it looks much more interesting, will
screenshot it

chrishamm, 2019-04-11:
Will be fixed in the next version.

chrishamm, 2019-05-14:
Fixed in RC7.

closed 2019-05-14 by chrishamm