Duet3D/DuetWebControl #130: Current Temperature shown as n/a and unable to go back to Tools

thedevleon, 2019-04-06:
I built the latest Duet Web Control 2.0.0-RC6+3, but also tried 2.0.0-RC6, happens on both.

1. For some reason, in the Tool page, all the Current temperatures just read as n/a. The graph next to it shows all temperatures fine though.

[image]

2. If I click on Extra, any clicks on Tools will simply not do anything. I have no possibility to go back to Tools, other than to reload the page.

For 2), the console shows this:

app.806a5f0e.js:19 TypeError: Cannot read property 'current' of undefined
    at app.806a5f0e.js:32
    at s.Ve [as _l] (app.806a5f0e.js:19)
    at sp (app.806a5f0e.js:32)
    at s.t._render (app.806a5f0e.js:19)
    at s.i (app.806a5f0e.js:19)
    at ni.get (app.806a5f0e.js:19)
    at ni.run (app.806a5f0e.js:19)
    at Jn (app.806a5f0e.js:19)
    at Array.<anonymous> (app.806a5f0e.js:19)
    at ce (app.806a5f0e.js:19)

chrishamm, 2019-04-08:
Can you send me your config.g?

thedevleon, 2019-04-08:
Sure, here it is:

```
; General preferences
G90                                                ; Send absolute coordinates...
M83                                                ; ...but relative extruder moves

; Network
M550 P"ender3"                                     ; Set machine name
M552 S1                                            ; Enable network
M586 P0 S1                                         ; Enable HTTP
M586 P1 S0                                         ; Disable FTP
M586 P2 S0                                         ; Disable Telnet

; Drives
M569 P0 S0                                         ; Drive 0 goes backwards, X
M569 P1 S0                                         ; Drive 1 goes backwards, Y
M569 P2 S1                                         ; Drive 2 goes forwards, Z
M569 P3 S0                                         ; Drive 3 goes backwards, E1
M569 P4 S0                                         ; Drive 4 goes backwards, E2
M350 X16 Y16 Z16 E16:16 I1                         ; Configure microstepping with interpolation
M92 X160.00 Y160.00 Z400.00 E830.00:830.00         ; Set steps per mm
M566 X900.00 Y900.00 Z30.00 E2000.00:2000.00       ; Set maximum instantaneous speed changes (mm/min), jerk
M203 X18000.00 Y18000.00 Z600.00 E6000.00:6000.00  ; Set maximum speeds (mm/min)
M201 X6000.00 Y6000.00 Z300.00 E6000.00:6000.00    ; Set accelerations (mm/s^2)
M906 X1260.00 Y1260.00 Z725.00 E1000.50:1000.50 I30 ; Set motor currents (mA) and motor idle factor in per cent

M84 S30                                            ; Set idle timeout

; Speeds
M204 P1000 T4000                                   ; Set print and travel acceleration
M207 S4.0 F2400 Z0.075                             ; Set firmware retract

; Axis Limits
M208 X0 Y0 Z-5 S1                                   ; Set axis minima
M208 X253 Y234 Z250 S0                              ; Set axis maxima

; Endstops
M574 X1 Y1 S1                                      ; Set active high endstops

; Z-Probe
M574 Z1 S2                                         ; Set endstops controlled by probe
M307 H5 A-1 C-1 D-1                                ; Disable heater on PWM channel 5 for BLTouch
M558 P9 H5 F120 T6000                              ; Set Z probe type to bltouch and the dive height + speeds
G31 P500 X-61 Y-28 Z4.25                           ; Set Z probe trigger value, offset and trigger height
M557 X5:253 Y5:234 S50                             ; Define mesh grid

; Heaters 

; Bed
M305 P0 T100000 B3950 R4700                        ; Set thermistor + ADC parameters for heater 0
M143 H0 S120                                       ; Set temperature limit for heater 0 to 120C
M307 H0 A274.1 C549.7 D9.4 S1.00 V23.7 B0          ; PID for Bed

; Hotend
M305 P1 T100000 B4725 C7.06e-8 R4700               ; Set thermistor + ADC parameters for heater 1
M143 H1 S280                                       ; Set temperature limit for heater 1 to 280C
M307 H1 A662.5 C243.2 D8.7 S1.00 V24.0 B0          ; PID for Hotend
M570 H1 P20 T20                                    ; Relax heater faults

; Laser
M307 H3 A-1 C-1 D-1                                ; Disable Heater output on H3

; Other
M307 H4 A-1 C-1 D-1                                ; Disable Heater output on H4

; Fans
M106 P0 S0 I0 F500 H-1 C"Air"                      ; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off, swap cooling
M106 P1 S0 I0 F500 H1 T45 C"Hotend"                ; Set fan 1 value, PWM signal inversion and frequency. Thermostatic control is turned on
M106 P2 S1 I0 F500 H-1 C"LEDs"                     ; Set fan 2 value, PWM signal inversion and frequency. Thermostatic control is turned off

; Tools
M563 P0 S"Extruder #1" D0 H1                       ; Define tool 0
G10 P0 X0 Y-18 Z0                                   ; Set tool 0 axis offsets
G10 P0 R0 S0                                       ; Set initial tool 0 active and standby temperatures to 0C

M563 P1 S"Extruder #2" D1 H1                       ; Define tool 1
G10 P1 X0 Y-18 Z0                                    ; Set tool 1 axis offsets
G10 P1 R0 S0                                       ; Set initial tool 1 active and standby temperatures to 0C

M563 P2 S"Laser"                                   ; Define tool 2
G10 P2 X0 Y-18 Z0                                    ; Set tool 2 axis offsets

M563 P3 S"Other"                                   ; Define tool 3
G10 P3 X0 Y-18 Z0                                    ; Set tool 3 axis offsets

; Temperature Sensors
M305 P103 X400 T22 S"Temperature"
M305 P104 X450 T22 S"Humidity [%]"
```

chrishamm, 2019-04-11:
Thank you, I could reproduce this problem. It will be fixed in the next version.

chrishamm, 2019-05-14:
Fixed in RC7.

closed 2019-05-14 by chrishamm