Duet3D/DuetWebControl #135: Dangerous extrusions with extrusions button?

Stephan3, 2019-04-28:
Hi @chrishamm 

I noticed that the dashboard buttons switch to realative, make move and switch back to absolute.
If one has configured absolute extrusions by default, the Button wont work as ecxpected.
What do you think about making it same behavior as X/Y/Z, using the relative switch ?

Thanks 
Stephan

mtsgsd, 2019-04-29:
This would be a great help. Manual extrusions or retractions would are part of trouble-shooting, loading, retracting or any other non-printing sort of need. Therefore you can't count on someone only needing to do a single retract or extrusion. Nor can you count on them knowing to  manually set relative mode to cure this.

chrishamm, 2019-05-07:
What relative switch? RepRapFirmware only supports an instruction for absolute coordinates (e.g. G53 G1 X200) when in relative mode. There is no support for relative extrusion in the same line yet.

RepRapFirmware supports multiple independent G-code channels so this is no problem there. Do you only have a single G-code channel in Klipper?

Stephan3, 2019-05-07:
@chrishamm i think yes, not sure.
The x/y/z move buttons are for what i am catching from the urls commands like:
G91
G1 X100 Fwhatever
G90

whereas extrusion came for me like
G1 E100 Fwhatever

If there is a G53 inbetween i missed it can be easily adopted.

Stephan3, 2019-05-10:
@chrishamm i have to reopen this.
By pushing the extrude retract button dwc2 RC6 sends the following urls:
```
/rr_gcode?gcode=G1%20E100%20F3600
/rr_gcode?gcode=G1%20E-100%20F3600
```

i tripplecheched with tcpdump. So no M83 is issued for me.

chrishamm, 2019-05-13:
OK, I see no reason not to add back the stack+M83 codes to DWC2. Will change it in the next version.

Stephan3, 2019-05-13:
@chrishamm you are my hero. Thanks 👍

closed 2019-05-13 by Stephan3