Duet3D/DuetWebControl #138: [FEATURE REQUEST] Powering the printer on/off via TP-Link SmartPlug

E1ixir66, 2019-06-04:
Would it be possible to implement the possibility for DWC to send M80/M81 commands to a TP-LINK SMART PLUG, powering the printer on or off?
Octoprint has a plugin to do that, and it is really very usefull! I spoke to DC42 about this, and he says  he thinks it would be possible, but it would require some code to be added to DWC to control the Smart Switch. He founds some Python code at https://github.com/softScheck/tplink-smartplug that could be translated into Javascript for use in DWC.

Thank you very much for your attention.
PS: Pleass pardon my poor english.

chrishamm, 2019-06-25:
Well, the question is if the TPLink SmartPlug provides any CORS access. If it does not, then there is no point in adding support to DWC since requests to it would be blocked anyway from your browser.

chrishamm, 2019-06-25:
PS: An implementation would be possible with the upcoming Duet 3, though.

chrishamm, 2019-10-07:
I'm closing this because I don't see how it is supposed to work without the TPLink Smart Switch offering CORS headers. In case you're planning on upgrading to a Duet 3, a plugin via DSF could provide this kind of functionality.

closed 2019-10-07 by chrishamm