Duet3D/DuetWebControl #139: Stray clicks in editor may lead to a reboot and corrupt files

jbitterly, 2019-06-07:
Twice now I've gone to click "Save" when editing my config.g in the system editor, clicking roughly in the highlighted circle here:
[image]

and have accidentally double clicked instead of single clicked which leads to a click being registered here:
[image]

which reboots the Duet board while it is trying to write my changes and results in a corrupted config.g file and no boot.  Oops.  

I think it would be a little safer if the system editor were drawn below the status bar containing the Emergency Stop button, this way you can't accidentally click it when trying to save your file and as a bonus, the E-Stop is available even if you are making a modification to a file.

dandancheeseandham, 2019-06-07:
Ahhh. I'd wondered what had happened about a month ago, when I managed to save my config and it broke. It showed a 0B file afterwards...

jbitterly, 2019-06-08:
Yea, it took me a minute to figure out what I had done the first time.  Once I pulled the SD card and noticed the bad config file I was able to restore from the backup and reproduce the feat shown above.

chrishamm, 2019-10-08:
This will be fixed in DWC 2.0.1. Thanks for your report!

chrishamm, 2019-10-08:
Fixed in v2.0.1.

closed 2019-10-08 by chrishamm