Duet3D/DuetWebControl #140: Assets failing to load with a reverse proxy on latest RC

aequasi, 2019-06-08:
I'm trying to set up a reverse proxy for this. In the past, the asset links were relative, but now they seem to be absolute. Is there a way to set up a reverse proxy on a given path? e.g. https://aequasi.local/duet/

chrishamm, 2019-06-25:
Hmm, I think this is related to WebPack but I don't know of any way to omit the prefixing '/' of the paths in the generated HTML file. It may be sufficient to delete the leading slash in the main HTML file to get it working on your setup but I haven't tried that yet.

hmortensen83, 2020-04-24:
"publicPath: './'" in config does solve most of the problems, but not all, and makes quite a few new problems with the routes and script refs.
Easiest way to solve this is to use another port on the proxy like https://domain.tld:1234/

mydevpeeps, 2020-08-24:
@aequasi  did you ever get this working?

mydevpeeps, 2020-08-25:
I was able to setup a subdomain (in dns) pointing to the cloud on an apache server (using https) using a configuration like below and it works without issue. This is a much better option than using a uri sub-path.

Connection Path:
Client (Restricted IP) -> Cloud Apache Server acting as Proxy (with SSL via certbot/zerossl) -> Home network on odd port (no SSL, not 80) -> Home Proxy Server (IP Restricted from Cloud Apache Server) (receives on odd port) -> Duet Wifi board (on port 80). 

_Advice:_ Assuming your internal network is flat, use a naming convention that makes sense per printer. I chose to use the last octet of the ip, so for example, a printer on IP 192.168.1.12 would be duet12.domain.com or 3d12.domain.com.

Apache Config Example:
_In this example, 10.10.10.10 is the home network external firewall ip, 10.11.0.x are fake ips i put in this example to demonstrate being able to restrict access to this domain (in the cloud) to specific source ips Port 81 is an example of an odd port to use to come into your network (a lot of ISPs will have a router that prevents ports 80/443 from being routed through your endpoint). The rewrite stuff is for the SSL and isn't needed if you aren't using it._
`
<VirtualHost *:80>
    ServerAdmin admin@domain.com
    DocumentRoot "/var/www"
    ServerName duet12.domain.com
    ErrorLog "logs/duet.log"
    CustomLog "logs/duet.log" common
    UseCanonicalName on
    ProxyPreserveHost on
    <Proxy *>
        Order deny,allow
        Deny from all
        Allow from 10.11.0.1
        Allow from 10.11.0.2
        Allow from 10.11.0.3
    </Proxy>
    ProxyPass / http://10.10.10.10:81/
    ProxyPassReverse / http://10.10.10.10:81/
    <Directory /var/www/html>
        Options +FollowSymLinks -Indexes
        #AllowOverride Indexes
    </Directory>
RewriteEngine on
RewriteCond %{SERVER_NAME} =duet12.domain.com
RewriteRule ^ https://%{SERVER_NAME}%{REQUEST_URI} [END,NE,R=permanent]
</VirtualHost>

`

chrishamm, 2020-09-03:
Now that I've got some more experience with Vue/Webpack I can say this issue is a configuration problem at build-time. I won't change the defaults, which means you'll have to compile it yourself in order to put it into a different subdirectory. You will need to change [publicPath](https://cli.vuejs.org/config/#publicpath) in vue.config.js and then rebuild it using npm/yarn. That should allow you to load it from a different directory.

DWC's HTTP requests always go to the root (/) but I'll change v3.2 to respect the configured base directory in case it differs. I can't help with the configuration of your actual webserver, though.

closed 2020-09-03 by chrishamm