Duet3D/DuetWebControl #142: No RC7 in releases section of GitHub

ctilley83, 2019-06-10:
I hear people running RC7 in other forums.  Is it officially released? I don't see it in the releases section.

rafaljot, 2019-06-10:
https://github.com/chrishamm/DuetWebControl/tree/2.0.0-RC7

JohnOCFII, 2019-06-10:
Odd that it does not show up at the top of that list -- RC-6 would appear to be the latest unless you scroll down.

rafaljot, 2019-06-10:
[image]

ctilley83, 2019-06-10:
> https://github.com/chrishamm/DuetWebControl/tree/2.0.0-RC7

That's not it. The last commit to that repository was 7 moths ago.  The latest changes are in the "next" branch

ctilley83, 2019-06-10:
[image]

This is where it should be

cdedwards, 2019-06-11:
maybe you could run the npm install and npm run build commands after installing npm. in other words build it yourself.
[image]

chrishamm, 2019-06-11:
I think the RC7 tag pointed at the wrong branch so I just recreated it and updated the release. Should be all good now. Thanks for reporting it!

closed 2019-06-11 by chrishamm