Duet3D/DuetWebControl #146: T0 fan- Tool Fan always visible

GivvHub, 2019-06-23:
I changed name of P0 fan:
M106 P0 L0 X255 C"Cooling printout"
And its visible at main page, at the end, sections: fan control, but is visible default tool fan too. Shoud be one name for one fan

[image]

chrishamm, 2019-06-25:
Your report does not describe the tool fan correctly - the tool fan is the one that is assigned to the currently selected tool, which only defaults to fan 0 (see M563 F-parameter). I agree it would make sense to hide the corresponding selection box if no fan is selected or if the currently active tool does not have any fans assigned to it, though.

chrishamm, 2020-01-13:
DWC2 displays the "Tool Fan" only if the currently selected tool has at least one fan assigned. Hence I'm closing this.

closed 2020-01-13 by chrishamm