Duet3D/DuetWebControl #149: 2.0.0 RC7 - Can't right-click file in Mac Safari

JohnOCFII, 2019-07-01:
I just upgraded from RC6 to RC7 and realized I can't right-click files in the list to make a selection (such as to delete the file).

Rolling back to RC6 and the ability to right click, then select an action behaves as expected.

fundix, 2019-09-01:
I have Safari too, but I can click normaly. 
[image]

JohnOCFII, 2019-09-01:
Hmm...  I'll have to try RC7 again.

chrishamm, 2019-10-07:
Did that apply to Safari on OS X or on an iPad/iPhone?

JohnOCFII, 2019-10-07:
> Did that apply to Safari on OS X or on an iPad/iPhone?

That was OS X.  I'll have to update to the released version and check again.  Also, Safari has been updated since then as well.  I'm away from my printer for a couple of days, but hope to get to this Wednesday.

chrishamm, 2019-10-08:
I just tried it on OS X 10.14.2 and it works perfectly there. If this problem shows up again, please open a new issue with instructions on how to reproduce it.

closed 2019-10-08 by chrishamm