Duet3D/DuetWebControl #151: [Feature Request] Ability to have temps display in layer chart graph in job status

jrsphoto, 2019-07-24:
The Layer Chart is super nice to have but I was wondering if it would be possible to have extruder and bed temps display in the same graph, with temp labels on the right side?   Just a thought.

jrsphoto, 2019-07-24:
And yep, I am aware that the temps are on a graph of there own, It just seems that you could combine the two graphs, or at least give the end user the option to do that?

chrishamm, 2019-11-05:
Well, I guess it would possible to keep track of the heater temperatures whenever a layer change occurs.

jrsphoto, 2020-01-13:
Just saw this Chris.. Thanks for doing this!

chrishamm, 2022-01-29:
It looks like this was already implemented in a previous DWC version but not displayed in the chart. I'll fix this in the next version.

chrishamm, 2022-02-10:
Now implemented in 3.4-rc1.

closed 2022-02-10 by chrishamm