Duet3D/DuetWebControl #152: RC7 - Tool, extruder and speed all stay increased

tonypartridge, 2019-07-24:
When I increase the extruder, fan or speed on a print it works great. But, when I upload a new GCODE for printing it doesn’t reset them to defaults?

I would expect it too, because you might have increased it for the final print to be say 150% but would not want to start a print at this

tonypartridge, 2019-07-24:
Check this, just noticed it on checking my state. 

000000 speed factor

[image]

chrishamm, 2019-07-26:
Seeing 0000 in the slider tip isn't nice, I will fix that. Thanks for pointing it out.

To reset the values to your defaults I suggest you either add M0/M1 to your end G-code and edit stop.g/sleep.g accordingly or you add the corresponding instructions (M220/221) right to your end G-code.

If you are not happy with this, I suggest you open a new issue on the RepRapFirmware repository.

tonypartridge, 2019-07-30:
Great thanks, if that’s how DUET natively handle it it’s fine. Its just a little unexpected from a users point of view.

chrishamm, 2019-10-01:
Fixed in v2.0.0.

closed 2019-10-01 by chrishamm