Duet3D/DuetWebControl #153: Feature request: G-code preview

TheBum, 2019-07-29:
I'd like to see a G-code preview similar to the one in OctoPrint. It would help verify you have the G-code uploaded that you thought you did. My workaround now is to download the file from the Duet 2 (Maestro) and load it into Voxelizer, but downloads from the board are really slow.

swblum, 2019-07-29:
Have a look at this [https://github.com/Sindarius/DWC_GCode_Viewer](url)
It is a good start for what you are trying to accomplish.

TheBum, 2019-08-01:
I just tried Sindarius's fork of DWC (https://github.com/Sindarius/DuetWebControl) and it works rather well, albeit rather slow to render. Hopefully he'll create a pull request soon to include his changes in chrishamm's repo.

chrishamm, 2019-10-08:
I'd like to integrate it as a separate plugin but it's great to see there's already a fork with support for it! It's on my TODO list.

Sindarius, 2019-10-08:
Let me know what I can do to help this along.

chrishamm, 2020-01-15:
Just a short update: This is planned for DWC 2.1.0.

Sindarius, 2020-01-15:
> Just a short update: This is planned for DWC 2.1.0.

That's exciting. Looking forward to getting the viewer included as a plugin. I have a few features I am aiming to add to it soon.

TheBum, 2020-01-15:
Hopefully one of those features is a layer preview so that individual layers can be inspected.

DMCShep, 2020-01-15:
@chrishamm Er wait, is this referring to a GCode Preview or a plugin system?

> Just a short update: This is planned for DWC 2.1.0.

Sindarius, 2020-01-16:
Something different than the fork I worked I assume?

christophmuellerorg, 2020-04-01:
Would be amazing in combination with the new [object cancelation feature](https://github.com/dc42/RepRapFirmware/blob/v3-dev/WHATS_NEW_RRF3.md) just introduced in RRF. Each object could be coloured differently and allow for cancelation through a context menu...

RomanLyavin, 2020-04-28:
it is very cool if this is implemented. I will wait! Thank you for adding this to your plans.

chrishamm, 2021-02-07:
Implemented in v3.2, enable the corresponding plugin in Settings -> General -> Plugins.

closed 2021-02-07 by chrishamm