Duet3D/DuetWebControl #154: Feature Request - Push Notifications

tonypartridge, 2019-07-30:
What about adding in push notification support? Browsers can actively have this? I.e. visit Duet, then duet requests permissions to send the browser a notifications?

chrishamm, 2019-07-30:
DWC1 actually supports them but I have not considered adding them back to DWC2. Also, since the Duet does not provide an HTTPS connection by default, there is no way to implement a true background worker like other pages do.

Perhaps we should also define what kind of notifications could be displayed via push notifications before this is actually implemented in DWC2.

KrohmKoala, 2019-10-06:
Could be cool ! Having notification for basic things like print finish, filament change required, filament runout, errors, and maybe some others.

Sindarius, 2019-11-27:
I ran into this some time ago and made a tool that would poll the duet and provide updates to a Discord channel. This may be a holdover for you until this enhancement is complete. https://github.com/Sindarius/DuetDiscordNotification

micaheli, 2021-03-18:
I would love this.  https support would be wonderful as well, if for any reason, just so I can expose my duet to the internet and not have to worry about somebody snooping my password.   

Notifications would just be icing on the cake.  My printer idled hot all night one time because the filament sensor falsely notified.  Came back to a skirt and first layer.