Duet3D/DuetWebControl #157: 2.0.0-RC7: CSS is using a too large font for dropdown boxes

ikarisan, 2019-08-22:
Hi!

I am running the 2.0.0-RC7 version of the Duet Web Control on a 24" TFT. But my browser window is not in full screen mode because I like to have many dedicated windows open at the same time. As a result the dropdown boxes for the tool's temperatures are using a too large font. Especially the hotend temperatures are only displayed as "2" "2" "2" "1" "1" instead of "235" "230" 205" "160" 100"...

[image]

Maybe the whole temperature chart shoould be moved below the "Status" and the "Tools" frames.  This would help alot when using a low resolution tablet (like the Amazon Fire 7) to view the GUI.

chrishamm, 2019-10-01:
Can you check if the problem persists in the latest version?

LittleHobbyShop, 2019-10-05:
I have this same issue in 2.0.0 latest release.

FCosta1, 2019-10-05:
Mine is ok in the last version..
[image]

maciejzarnowski, 2019-10-07:
> 
> 
> Can you check if the problem persists in the latest version?

The problem is still there when there are more than 5 temperatures defined and the side slider is active.

chrishamm, 2019-10-07:
I'll make the tool temperature inputs wider in the next release and I hope that will fix it.

FCosta1, 2019-10-07:
[image]

Mine is correct...